todd-herbert/heltec-eink-modules #1: HELTEC 2.90" V2 - Multiple incompatible versions circulating

todd-herbert, 2021-12-01:
It has come to my attention that "not all displays are created equal". A new batch of 2.90" Black and White displays appears to have a different panel which is not currently supported by this library. This will be investigated in the near future.

todd-herbert, 2021-12-06:
Preliminary support for the new panel has been added in ce931b6c1bd68e7551cde5f1f1f169fcbb8e0df1
Expect partial refresh support, examples and documentation to follow shortly.

todd-herbert, 2022-02-16:
Further investigation has revealed that while this new panel does support partial refresh, it is not possible to implement this with paging.

timstrasser, 2023-02-17:
> Further investigation has revealed that while this new panel does support partial refresh, it is not possible to implement this with paging.

Might this exact issue also cause problems with the 152x152 1.54" displays?

todd-herbert, 2023-05-02:
> Further investigation has revealed that while this new panel does support partial refresh, it is not possible to implement this with paging.

I was mistaken here. I had misunderstood how partial refresh should be performed for this display.
*The model is **HTE029A1**. Support has been added in the [dev branch](/todd-herbert/heltec-eink-modules/tree/dev), and will be incorporated in a future release.

todd-herbert, 2023-05-06:
Display supported as of [v2.1.0](/todd-herbert/heltec-eink-modules/releases/tag/v2.1.0)

closed 2023-05-06 by todd-herbert