todd-herbert/heltec-eink-modules #2: Heltec 1.54" V2 display incompatible

timstrasser, 2023-02-17:
I'm trying to use this library for my 1.54" V2 display, but when running the examples it all looks from. After playing around for a bit, I found that my display has a resolution of 152x152 and not 200x200, as stated here: https://heltec.org/project/154-e-ink/.

After I changed the resolution in the code, the display seems to work somewhat, but only some of it. Some display updates still produce garbage and I got some weird lines on the right side of the display.

Right now, I'm a bit confused as why the resolution in the code is set to 200x200, when the specs say 152x152. Even the AliExpress listing I bought my display from claims it to be 200x200.

todd-herbert, 2023-02-18:
Unfortunately, it does seem that Heltec have released yet another wave of non-interchangeable displays. If there are any visual differences between the two models, they're likely to be very subtle.

***

I looked into this all for you today, and I can confirm that the display I was using for development does work with the library. I then had a look through my spare parts and found another new "1.54-inch V2". It had a few small differences in appearance; most noticeable is the blue tab on the screen protector. This new display did not respond at all to this (heltec-eink-modules) library.
> This library (https://github.com/HelTecAutomation/e-ink) does work with my display, 

It also failed to respond to the official Heltec library, where you had some success.
In the end, it appears to work with [GxEPD2](https://github.com/ZinggJM/GxEPD2), as `GDEP015OC1`.

***

As far as I can tell, Heltec has marketed at least three (non-interchangable) panels as the same "1.54 Inch E-ink Display V2"
| Model ( w/ datasheet) | Resolution | Introduced | Status | Currently Supported By |
| ------------------------ | ------------ | ------------ | ------- | --------------- |
| [GDEP015OC1](https://github.com/todd-herbert/heltec-eink-modules/files/10774202/GDEP015OC1.pdf)              | 200 x 200 | before 2020| No longer in production | GxEPD
| [DEPG0150BNS810F0](https://github.com/todd-herbert/heltec-eink-modules/files/10774199/DEPG0150BNS810F0.pdf)   | 200 x 200 | 2020 | In production (?) | This Library, GxEPD, Official Heltec |
| [DEPG0154BNS800F35](https://github.com/todd-herbert/heltec-eink-modules/files/10774200/DEPG0154BNS800F35.pdf) | 152 x 152 | 2021 | In production | Official Heltec |

***

I first threw this library together because it seemed odd that there was no neat and tidy, out of the box, solution to work with these Heltec displays. At that time, I assumed that it was a small and stable range of products, but it's looking like that may not be the case.

Unfortunately, this library doesn't support your panel right now  (`DEPG0154BNS800F35`). To tell you the truth, I had no idea that a revision was being sold. If you are able to link me to the vendor you got your display from, I will look into adding support, but it could be some time away, even just with the shipping.

I don't believe that this panel is supported currently by GxEPD2 either, however I would encourage you to look into that further. The Heltec Official Library is really more of a tech demo. Its biggest limitation is that it does not implement [paging](https://github.com/todd-herbert/heltec-eink-modules#what-is-paging), so an Arduino UNO will likely not have enough memory to render graphics at runtime. If you are using a different microcontroller, this may not be an issue.

timstrasser, 2023-02-18:
Thanks for your answer.
My display has a red tab on the screen protector. I got my display from this AliExpress listing: https://www.aliexpress.com/item/4000560969424.html. After a bit more digging, it seems like the problem is that this display used the IL0373 driver, and there's something weird about its resolution. The AliExpress listing states 200x200, but that's clearly wrong, then the datasheet lists 152x152, but then's an 8 pixel gap, so 160x160 seems to do the trick (but then there's still clipping).

With the answer from this post (https://forum.arduino.cc/t/gxepd2-unable-to-connect-esp32-c3-with-1-54inch-e-ink-display-il3829/1039682/10), I now managed to get GxEPD2 to a somewhat workable state. Images and text is still clipping left and right, but at least I get the display to show something recognizable.

I used this (https://github.com/ZinggJM/GxEPD2/compare/master...TheRoam:GxEPD2_Heltec-mods:master) modded GxEPD2 library for that. Would it be possible to use this as a starting point to integrate this display variation into this library as well?

todd-herbert, 2023-02-18:
>  The AliExpress listing states 200x200, but that's clearly wrong, then the datasheet lists 152x152

The sellers seem to be pretty much unaware that the product they are stocking has changed. Can't really blame them though, when it has the same model number printed on it.

> I now managed to get GxEPD2 to a somewhat workable state. Images and text is still clipping left and right, but at least I get the display to show something recognizable.

Ah, I do remember running into something like this with other displays. There was some quirk that the panel doesn't really draw the first byte of memory for each row. Before the image data is sent, the library needs to send a set of SPI commands to "Specify the start/end positions of the window address". This might mean that you're going to lose 8 pixels horizontally, but possibly not.

I don't know GxEPD2, but I would see if the `display.setPartialWindow(box_x, box_y, box_w, box_h)` method does the trick.
I'd go for 
- `x=8; y=0; w=152-8; h=152`
- `x=8; y=0; w=152; h=152`

or something to this effect

todd-herbert, 2023-02-18:
Oh wait one minute,

> With the answer from this post (https://forum.arduino.cc/t/gxepd2-unable-to-connect-esp32-c3-with-1-54inch-e-ink-display-il3829/1039682/10),

If this is the case, it looks like you have an older panel rather than a newer model. I didn't realize this one was around. You'll want the datasheet for [HTEW0154T8_V2](https://resource.heltec.cn/download/e-ink/154/1.54b%26w/HTEW0154T8_V20/HTEW0154T8_V20.pdf)

timstrasser, 2023-02-18:
> You'll want the datasheet for [HTEW0154T8_V2](https://resource.heltec.cn/download/e-ink/154/1.54b%26w/HTEW0154T8_V20/HTEW0154T8_V20.pdf)

I've really lost track of all the different yet same displays there are at this point. Which library / settings would you think could work best for this display?

timstrasser, 2023-02-18:
> I don't know GxEPD2, but I would see if the `display.setPartialWindow(box_x, box_y, box_w, box_h)` method does the trick.

I tried to add this to the drawFont method in the GxEPD2 Example, but it does not seem to make any difference.
For me, the text is still clipped on both sides.

```cpp
void drawFont(const char name[], const GFXfont* f)
{
  display.setPartialWindow(8, 0, 152, 152);
  //display.setRotation(0);
  display.fillScreen(GxEPD_WHITE);
  display.setTextColor(GxEPD_BLACK);
 
  display.setFont(f);
  display.setCursor(0, 0);
  display.println();
  display.println(name);
  display.println(" !\"#$%&'()*+,-./");
  display.println("0123456789:;<=>?");
  display.println("@ABCDEFGHIJKLMNO");
  display.println("PQRSTUVWXYZ[\\]^_");
  if (display.epd2.hasColor)
  {
    display.setTextColor(GxEPD_RED);
  }
  display.println("`abcdefghijklmno");
  display.println("pqrstuvwxyz{|}~ ");
}
```

todd-herbert, 2023-02-18:
I've had a quick look at the Official Heltec example for that display, and it looks like they might be doing the equivalent of `display.setPartialWindow(0, 0, 144, 152);`
You may or may not need to change the defined size of your panel too. Any chance you can post a photo of what your display is doing?

I don't know too much about GxEPD2, but from what I saw in that thread, the code wasn't specifically for `HTEW0154T8_V2`, it seemed to be just generic code that the panel was more or less going with. If you want to go further with it, you might need to start fishing out parts from the working good example at HelTecAutomation/e-ink and slotting them in to GxEPD2.

todd-herbert, 2023-02-18:
> > You'll want the datasheet for [HTEW0154T8_V2](https://resource.heltec.cn/download/e-ink/154/1.54b%26w/HTEW0154T8_V20/HTEW0154T8_V20.pdf)
> 
> I've really lost track of all the different yet same displays there are at this point. Which library / settings would you think could work best for this display?

This was an even earlier display I believe. The panel "appears" to be made by Heltec themselves, so unlike the newer generic designs, there doesn't seem to be much third party support. My guess would also be that the model is no longer manufactured, only available as limited old stock.

I will definitely look into writing code for it, but at this point you might be pretty much on your own, either fitting it in to GxEPD2, or rewriting the official Heltec Library example to suit your needs.

timstrasser, 2023-02-18:
> Any chance you can post a photo of what your display is doing?

Yes, sure. With `display.setPartialWindow(8, 0, 144, 152);`, [this](https://share.unkn0wn.website/screenshots/j/55gyXEFh0WVcPMr) is what it looks like. It seems like the width was limited by the partial window (before, the text continued to the right border), but it seems like the x param does not change anything. If it weren’t for this clipping on all sides, I think GxEPD2 would be usable with this display.

I'm very glad to hear, you're planning on supporting this display in the future. I'm in no rush. If there's anything I can do to help, let me know. I'd be happy to help testing.

todd-herbert, 2023-02-18:
You might be better off removing that line of setPartialWindow(), and going straight for the source, and adjust the GxEPD2_160_BN.cpp file:
 ```cpp
void GxEPD2_160_BN::_setPartialRamArea(uint16_t x, uint16_t y, uint16_t w, uint16_t h)
{

  x += 8;
  y += 0;
  w += 0;
  h += 0;

  // _writeCommand(0x11); // set ram entry mode
  // _writeData(0x03);    // x increase, y increase : normal mode
  _writeCommand(0x44);
  _writeData(x / 8);
  _writeData((x + w - 1) / 8);
  _writeCommand(0x45);
  _writeData(y % 256);
  _writeData(y / 256);
  _writeData((y + h - 1) % 256);
  _writeData((y + h - 1) / 256);
  _writeCommand(0x4e);
  _writeData(x / 8);
  _writeCommand(0x4f);
  _writeData(y % 256);
  _writeData(y / 256);
}
```

I've changed the file here, if it makes it clearer: https://github.com/todd-herbert/GxEPD2_Heltec-mods

timstrasser, 2023-02-18:
I've played around for a bit more, and figured out, that calling setPartialWindow() does not actually fix anything, it just narrows down the part of the display that's actually redrawn. However, it seems like in my case that's not the issue.

I also tried your modification, but it seems to make things way worse. I'm not sure why, but with that modification, the text does not seem to display any more (I could only briefly catch a glimpse of what could be a character).

Currently, the workaround for me is adding 8 to the x-axis anytime I draw something. E.g. I can print text by disable word wrap and calling setCursor(8, 16*lineNumber). I won't be able to use automatic line breaks (because the next line would start at x=0 again), but I think I can live with that for now.

This way, the text looks alright and with this the display is now more usable than it ever was with the official library.

todd-herbert, 2023-02-18:
It's good to hear you've got something working. Without a display here to test, I'm definitely only guessing. If you have the motivation, I would suggest tinkering with that _setPartialRamArea() method; the right set of parameters there will probably fix the issue. Other than that, I'll let you know if I get any further.

timstrasser, 2023-02-18:
Alright, I appreciate that. I'll definitely play around with it some more as soon as I have some more time at hand. I will let you know when I figure out anything. For now, though, it will have to wait, as I've already spent too much time that I don’t have. 😬

Looking forward to continuing working on this in the future.

timstrasser, 2023-02-20:
So, I played around for a bit. My mystery is still unsolved, and even more questions have appeared.
This code produces two perfect diagonals across the screen:
```cpp
display.drawLine(8, 0, 160, 152, GxEPD_BLACK);
display.drawLine(160, 0, 8, 152, GxEPD_BLACK);
```
So, it seems like the entire canvas is shifted to the left by 8 pixels. This could be caused by the 160x160 controller of the display.

To fix it, I added 8 to the x coordinate in `GxEPD2_160_BN::_setPartialRamArea()` and indeed, this time it seemed to work. This time, the following code produces two perfect diagonals:
```cpp
display.drawLine(0, 0, 152, 152, GxEPD_BLACK);
display.drawLine(152, 0, 0, 152, GxEPD_BLACK);
```

However, it took not long to discover that now other things broke.
Before my change, this drew a bitmap perfectly across the entire screen:
```cpp
display.drawBitmap(0, 0, myBitmap, 152, 152, GxEPD_WHITE);
```
With the change, however, the bitmap is now also shifted to the right. Also, methods like `display.fillScreen(GxEPD_WHITE);` no longer work correctly, as they also seem to be shifted to the right and no longer fill the leftmost 8 pixels.

The thing I don't understand: Why would `display.drawLine()` now work as expected, but `display.drawBitmap()` not any more, when they both are methods of the Adafruit GFX class? I'd expect them to function similarly.

todd-herbert, 2023-02-21:
Sounding good so far, any luck with the fast "partial refresh"?

> Before my change, this drew a bitmap perfectly across the entire screen

Purely for convenience, you might want to look into the drawXBitmap method. The xbm format is a plain text, C array of pixels data that you can just about ```#import "my_image.xbm"``` and feed straight into the method. GIMP supports the format.

[Preparing Mono XBitmaps for E-INK](https://github.com/todd-herbert/heltec-eink-modules/blob/ba082db208f8d2bf211149699d63fed3c2916d4b/docs/XBitmapTutorial/mono.md)

timstrasser, 2023-02-21:
> Sounding good so far, any luck with the fast "partial refresh"?

What do you mean exactly by fast partial refresh?
Do you mean something like this? https://youtube.com/shorts/Zn6ws0cfpkk

todd-herbert, 2023-03-01:
> Do you mean something like this? https://youtube.com/shorts/Zn6ws0cfpkk
Yep. Is that video of your panel? It seems to be running quite well.

timstrasser, 2023-03-01:
Yes, that's my Heltec panel. Everything except line breaks seems to work now. (The line breaks a little bit to late, because the library still thinks it's a 160x160 display.

This is completely offtopic now, but (out of frustration with the Heltec display), I bought another display. This time from MH-ET LIVE, because in the AliExpress reviews it looked like that one should work with GxEPD2 right out of the box. But of course this can't be the case. I'm now a few hours into trying to get the display to do something, but I did not have any luck so far.

So, the display that I got was the 2 color version (black/white/red), [1.54" MH-ET LIVE display](https://www.aliexpress.com/item/32880102057.html). From this [YouTube Video](https://www.youtube.com/watch?v=vHMCj7vwRc0) from the GoodDisplay channel, I then found the exact panel I received by comparing the numberings on the flex cable. So, I think it's the GDEH0154Z90 panel.

With that knowledge, I opened up the GxEPD2 Example and chose the GxEPD2_154_Z90c (I also tried GxEPD2_154c, which is the only other 3-color 200x200 option), connected everything 1000 times, tried different boards, but I can't get a single pixel to move.

I don't have any more ideas that I could try. There does not seem to be any official stuff (except a corrupt zip file from their repo, of which I don't even know if it's actually for my display).

On the back side of the panel, I noticed that there's an unpopulated spot, but I doubt it has anything to do with my problem.

Do you have any more ideas what I could try?

timstrasser, 2023-03-01:
Update: Got my problem with the new display sorted, for some weird reason, you cannot use the D8 pin nor the D6 pin for CS, but when you use D4 it works without a problem.

todd-herbert, 2023-04-27:
As best I understand, the display described in this issue is in fact [```DEPG0154BNS800F35```](https://github.com/todd-herbert/heltec-eink-modules/files/10774200/DEPG0154BNS800F35.pdf).

A development build for this display is now available on branch [feature/DEPG0154BNS800F35](/todd-herbert/heltec-eink-modules/tree/feature/DEPG0154BNS800F35).  
Documentation and examples may not yet reference this display.

A stable version will be bundled with the next release.

todd-herbert, 2023-05-06:
Display supported as of [v2.1.0](/todd-herbert/heltec-eink-modules/releases/tag/v2.1.0)

closed 2023-05-06 by todd-herbert