todd-herbert/heltec-eink-modules #6: ESP8266 broken with v3.0.0. With old dev, only black is broken.

RubenNL, 2023-09-12:
ESP8266: D1 mini  
display: DEPG0213RWS800

**Results with the last release, v3.0.0:**

The display doesn't update at all. Nothing happens.

**Results with commit [18e3aa](https://github.com/todd-herbert/heltec-eink-modules/commit/18e3aafcf9c619701427a1ca24fb65fa5080c43a):**

I can't get the colors to function like expected. Black and Red both give me a red color.

I've recorded a refresh of the old commit, where it looks like the circles are drawn at different stages.  
At the beginning of the refresh, it draws a black rectangle around the display, so i don't think my screen is broken. but: I don't have a Arduino to verify this.

https://github.com/todd-herbert/heltec-eink-modules/assets/15885455/64373b0b-8f48-4f76-8bcd-a960fe14f5d6

**Code:**

I used the same code with v3.0.0 and the last dev commit.
 
```c++
#include "heltec-eink-modules.h"
#define PIN_BUSY    5
#define PIN_CS      4
#define PIN_DC      2
DEPG0213RWS800 display( PIN_DC, PIN_CS, PIN_BUSY );      // 2.13" V2 - BWR - Red Tab
void setup() {
  display.fillCircle(75,50,50,RED);
  display.fillCircle(75,200,50,BLACK);
  display.overwrite();
}
void loop() {}
```

todd-herbert, 2023-09-12:
That is odd, thank you for letting me know!

I have just run your code on NodeMcu V3, and it appears to draw correctly. I have attached the result

* Would you be able to send images of your wiring?
* Could you confirm the text printed on the orange "flex connector" of your display?

It'd be good to double check these two things, just in case.

Other than that, I'll do a bit of research and see if I can come up with anything.

https://github.com/todd-herbert/heltec-eink-modules/assets/24772776/f6f39c21-f3de-4e2b-9b06-ee83d91d63fa

todd-herbert, 2023-09-12:
>The display.overwrite() call doesn't seem to be supported with v3.0.0.

Oh gosh.. It looks like this is the unexpected result of a last minute fix a8f817d
I will try get a patch out shortly

RubenNL, 2023-09-13:
Flex connector text: FPC-7528B. This is the same as in the readme `DEPG0213RWS800`.

I re-checked the wiring, and it seems to be okay. After re-wiring everything I still get the same results.

I suspect the controller or the screen to be broken. Where yours is fully flashing black, mine only has the border black.

I will try to get another display like this, but if that doesn't work out I will close this issue because I can't rule out defects of the hardware.

todd-herbert, 2023-09-13:
> I will try to get another display like this, but if that doesn't work out I will close this issue because I can't rule out defects of the hardware.

Unfortunately, I think you might be right, whether it's hardware, or a defect in the factory programming.

I tried to misusing a panel in a variety of ways, but I couldn't get a result similar to yours.
* Low voltage
* Restricted current
* Old commit
* Incorrect config - Display Update Control 1 ( Command 0x21)
* Tested with QYEG0213RWS800
* Tested new DEPG0213RWS800 from packet
* Tested with wrong class
* Tested with no init code

> Flex connector text: FPC-7528B. This is the same as in the readme DEPG0213RWS800.

As far as I am aware, `DEPG0213RWS800` is the only Heltec display with the label FPC-7528.
I have heard of the same label appearing on other displays, marketed by other companies. 
It is not *impossible* that your display is yet another undocumented variant.

Would you be able to tell me where your display came from? I'd like to keep an eye on the situation.

todd-herbert, 2023-09-27:
[image]

I haven't been able to reproduce the issue on a D1 mini. It is looking more like an issue with the panel.

closed 2025-04-22 by todd-herbert