todd-herbert/heltec-eink-modules #8: Refreshing the display in fast mode.

Wirewrap, 2023-12-22:
Totally new to the epaper display world.
I am running an instrumentation display, ESP32 and DEPG0290BNS75A.
The full display area is updated in fast mode every 10 seconds.

Would this snippet be enough to refresh the display? `refreshNeeded` is set to several  minutes, maybe 10.  
```
if (millis() - lastRefresh > refreshNeeded)
{
   display.fastmodeOff();
   display.update();
   display.fastmodeOn();
   lastRefresh = millis();
}

```
When I test this it still looks good after a day, no visible ghosting.

todd-herbert, 2023-12-23:
The term "refresh" is maybe one to be avoided here, as it is the technical term for the process of "updating the display image". Am I correct in understanding that the aim here is to preserve the display's health?

Unfortunately, I don't feel confident enough to give you a concrete answer about this, but I will pass on my general understanding:

When the display is updated, an electric charge attracts or repels the black pigment suspended inside the E-Ink display. If the display is repeatedly updated with the same image, some part of the system (too technical for me) will begin to retain its charge.

When the display is updated with fastmode off (technically "full refresh"), all pixels are "blinked" several times. When the display is updated with fastmode on ("partial refresh"), there is no blinking. The display, referring to its internal memory of the previous image, forcefully "pulls" the pigment in the correct direction, where required.

My understanding is that this gradual build up of charge, along with the imprecision of this "partial refresh" technique, can cause ghosting (burn-in?) to develop.

___

> When I test this it still looks good after a day, no visible ghosting.

If you continue to write to the same area, most of the after-image will be concealed. The true extent would be more apparent if you were to call `display.clear()` and observe a "blank" screen.

___

> Would this snippet be enough to refresh the display?

I would hope so. I can quote this advice from [waveshare](https://www.waveshare.com/wiki/4.2inch_e-Paper_Module_(B)_Manual):

  > For E-paper displays that support partial refresh, please note that you cannot refresh them with the partial refresh mode all the time. After refreshing partially several times, you need to fully refresh EPD once. Otherwise, the display effect will be abnormal, which cannot be repaired!

During the process of a "full refresh" (update with fastmode off) all pixels are toggled several times, giving a blinking effect. If you see concerning ghosting beginning to develop, performing this "full refresh" more often may be helpful.

Calling `display.clear()` will have the same restorative effect as updating with fastmode off. It can be called several times in a row, if even more of this "blinking" is required.

If you are not doing so already, I highly encourage you to **minimize fastmode updates by only redrawing the screen when necessary**: confirm that your data has actually changed before updating.
___

I should make it clear that the "partial refresh" operation (fastmode) for `DEPG0290BNS75A` has not come from any official source. With my limited knowledge I have kludged it together, so some caution is advised.

It may very much be a case of "see how it goes". In my limited experience, I haven't yet encountered a case of ghosting that wasn't reversible. That being said, I haven't tested days and weeks of continuous fastmode updates, so definitely keep an eye on it.

I have had success restoring displays with significant burn-in, by repeatedly drawing random noise for an extended period. While this has worked for me in the past, I would proceed with caution, as I have no idea about the impact on the display life-span.

Wirewrap, 2023-12-23:
Thank you for your thorough answer. I will proceed and see how it goes.

>The term "refresh" is maybe one to be avoided here, as it is the technical term for the process of "updating the display image". Am I correct in understanding that the aim here is to preserve the display's health?

Yes, "rejuvenate" was the term concerning B/W CRT tube, that is what I meant.   

> If you are not doing so already, I highly encourage you to minimize fastmode updates by only redrawing the screen when 
 necessary: confirm that your data has actually changed before updating.

That is definitely my plan. The display will show Oil temperature, Oil pressure and Fuel pressure. They will not change much over time

> I have had success restoring displays with significant burn-in, by repeatedly drawing random noise for an extended period. While this has worked for me in the past, I would proceed with caution, as I have no idea about the impact on the display life-span.

I will include that function in the service/setup menu. A couple of hundred hours of power on lifetime will be enough. Certainly less than 1000.

closed 2023-12-23 by Wirewrap