todd-herbert/heltec-eink-modules #11: Partial refresh inbetween deep sleeps

allllo20, 2024-03-05:
First of all, thank you so much for this project! It turned e-waste into something usable..

I have a specific problem that I tried in various ways to approach but none seem to work. I'm trying to code a clock with an esp32 that updates the time on the e-paper display after every minute and then goes to deep sleep for the rest of the time. The idea is to do a full update every full hour and the rest of the time should be partial (or fast-)updates. 

The GxEPD2 library allows to skip the initial refresh of the display when initializing. I can't find anything similar here. I basically tried the "fast_mode" example but added a deep sleep command like this:

`

void setup() {
    display.setTextSize(2);
    display.fastmodeOn();

    DRAW (display) {
        display.setCursor(0, f.bottom() - 30);
        display.println("Fastmode:");
        display.println("On");
    }

    display.setTextColor(WHITE);
    display.setWindow( f.left(), f.top(), f.width(), f.height() - 35 ); // Don't overwrite the bottom 35px

    for (int demo = 0; demo <= 5; demo++) { // Count up to 5

        DRAW (display) {
            // Draw the next "loading icon" in sequence, from hourglasses[]
            display.drawXBitmap(ICON_L, ICON_T, hourglasses[demo % 3], hourglass_1_width, hourglass_1_height, BLACK);

            // Draw a square in the corner with a digit
            display.fillRect(0, 0, 30, 30, BLACK);
            display.setCursor(10, 10);
            display.print(demo);
        }

    }
    display.fastmodeOff();

    esp_sleep_enable_timer_wakeup(2*1000000ULL);
    esp_deep_sleep_start();
}
`

todd-herbert, 2024-03-05:
Ah yes, I know what you mean.
The fact that the screen clears after a reboot was originally a deliberate design decision. I always figured a bunch of people would reboot with an existing image on the screen, and then end up with it smearing together with the first fast-mode update.

I hadn't actually considered a situation where the device is waking from deep-sleep, but it's definitely an important use case. It would be nice to disable the clear-at-boot behavior if the library can detect that a device *is* waking from sleep, but this might be a bit fiddly to implement across all the different platforms. I will probably look into though.

For now, it shouldn't be hard to add an optional argument to bypass the clear-at-boot- behavior. I should be able to get this out in the next few hours.

todd-herbert, 2024-03-06:
Hmm, okay. I've had a look at the library code. Apparently only devices with limited RAM, such as Arduino Uno, actually enforce a full-refresh clear at boot. With ESP32, no refresh takes place, but the library *does wipe the memory*.

I *will* get an update out shortly, which allows bypass of this initial memory wipe.

 ___

If I remember, this memory wipe is done to clear the white-noise pattern which fills the display memory if power is removed for >10 seconds. This memory loss is problematic for fast-mode, due to what goes on behind-the-scenes. 

For a normal refresh, no problem: the display just blinks and flashes until all the pixels are wrangled into the right spot. For fast-mode though, the display hardware needs to keep track of two sets of data: the new image for update, and the existing image on screen. When the update happens, the display only moves pixels which have changed between the old and new image. It is really a "differential update".

If the display loses power for an extended period (>10 seconds), the memory will be lost, and the display will believe that the "existing image" is just a white-noise pattern. It will be unable to move the correct pixels for a fast-mode update. 

**I mention this because I see you are trying to minimize power usage with `esp_deep_sleep_start()`.** 
I *believe* the displays have quite a low current-draw at standby, however this library does not put them into a true "deep-sleep" state. The reason is that Heltec did not break out the display's reset pin, so the only way to exit the "deep-sleep" state is to completely remove power from the display.

The library *does* have code to support this, however it is not currently suitable for your use-case. The [custom power switch](https://github.com/todd-herbert/heltec-eink-modules/blob/main/docs/Wiring/wiring_esp32.md#optional-suggested-additional-wiring-for-power-saving) feature is designed to remove all power from the display. It was not written with the intention of using the display's "deep-sleep" method. Because extended power-removal causes display memory loss, it is not possible to use fast-mode immediately after `customPowerOn()`.

What I *can* look into is writing separate methods to allow the same transistor set-up to use deep-sleep, and then a short power-down (>4s, <10s) to escape the deep-sleep mode. This *might* allow fast-mode directly after deep-sleep.

Let me know if you think this last option might be helpful, and I can prioritize exploring.

todd-herbert, 2024-03-06:
> What I can look into is writing separate methods to allow the same transistor set-up to use deep-sleep, and then a short power-down (>4s, <10s) to escape the deep-sleep mode. This might allow fast-mode directly after deep-sleep.

I experimented with this today, but there was no appreciable power-saving. Without a physical RST pin, I don't think these Heltec displays can really make use of their deep-sleep function.

todd-herbert, 2024-03-06:
Using fast-mode with ESP32 deep-sleep is now supported as of v4.2.0. You can install this right now by downloading the .zip release from github, or in the next few hours via the Arduino Library Manager.

You should pass the argument `fastmodeOn(false)` to bypass the initial memory clearing.

Note that the code in the 1st post would also need a call to `clearMemory()` at the start of each for loop. 
By design, `update()` does not erase your working image, allowing you to add to it by drawing `BLACK`, or erase by drawing `WHITE`. Calling `clearMemory()` will return the working image to the background color, before drawing the new and different hourglass graphic.

If you are not already doing so, I would suggest calling `clear()` at first boot, to make sure both the display and the memory are blank.

```cpp
void setup() {
    if (esp_sleep_get_wakeup_cause() != ESP_SLEEP_WAKEUP_TIMER)
        display.clear();

    display.setTextSize(2);
    display.fastmodeOn(false);  // false to disable memory-init

    display.setCursor(0, f.bottom() - 30);
    display.println("Fastmode:");
    display.println("On");
    display.update();

    display.setTextColor(WHITE);
    display.setWindow( f.left(), f.top(), f.width(), f.height() - 35 ); // Don't overwrite the bottom 35px

    for (int demo = 0; demo <= 5; demo++) { // Count up to 5
      display.clearMemory();    // Start with a blank image
      display.drawXBitmap(ICON_L, ICON_T, hourglasses[demo % 3], hourglass_1_width, hourglass_1_height, BLACK);
      display.fillRect(0, 0, 30, 30, BLACK);
      display.setCursor(10, 10);
      display.print(demo);
      display.update();
    }
    display.fastmodeOff();

    esp_sleep_enable_timer_wakeup(15*1000000ULL);
    esp_deep_sleep_start();
}
```

Thanks for taking the time to point out the issue. Let me know how you get on!

closed 2024-03-06 by todd-herbert

allllo20, 2024-03-22:
@todd-herbert  hey there again! I wasn't able to check back until today after the issue, but man.. thank you so much for the extensive explanation and blazing fast fix! Everything's working now as expected and I can't thank you enough. Is there a way I can spend you a coffee via PayPal or anything?

todd-herbert, 2024-03-23:
> Is there a way I can spend you a coffee via PayPal or anything?

No need, but thank you! Just glad to see you get the hardware working.

bohnelang, 2025-05-04:
Thanks you for your powerful project and the very useful examples. I am using an Heltec Master Vision E290 module for monitoring my PV inverter. I am using the deepsleep mode and the very first clearing with the flickering screen was not nice. With your libreray and this following example I figured out that a fastclear ist possible.

On boot I do have an initial full reset of the screen and every hour once. In between I use a "fastclear" function.  

To provide a short example I reduced my code to the following lines. I am not sure if I use the clear.memory function in the rigth way. May I ask for a code review of the void display_fastclear(void) part?

```
#include <heltec-eink-modules.h>
// Pick your panel  -  https://github.com/todd-herbert/heltec-eink-modules
// ---------------

    // "All-in-one" boards
    // --------------------------------------

        // EInkDisplay_WirelessPaperV1 display;
        // EInkDisplay_WirelessPaperV1_1 display
        // EInkDisplay_VisionMasterE213 display;
        EInkDisplay_VisionMasterE290 display;

    // SPI Displays
    // --------------------------------------

        // Wiring (SPI Displays only)
        #define PIN_DC   2
        #define PIN_CS   4
        #define PIN_BUSY 5

        // DEPG0150BNS810 display( PIN_DC, PIN_CS, PIN_BUSY );      // 1.54" - Mono 
        // DEPG0154BNS800 display( PIN_DC, PIN_CS, PIN_BUSY);       // 1.54" - Mono 
        // GDEP015OC1 display( PIN_DC, PIN_CS, PIN_BUSY);           // 1.54" - Mono 
        // DEPG0213RWS800 display( PIN_DC, PIN_CS, PIN_BUSY );      // 2.13" - 3 Color Red
        // QYEG0213RWS800 display( PIN_DC, PIN_CS, PIN_BUSY );      // 2.13" - 3 Color Red
        // DEPG0290BNS75A display( PIN_DC, PIN_CS, PIN_BUSY );      // 2.9"  - Mono 
        // DEPG0290BNS800 display( PIN_DC, PIN_CS, PIN_BUSY );      // 2.9"  - Mono 
        // GDE029A1 display( PIN_DC, PIN_CS, PIN_BUSY );            // 2.9"  - Mono 

/* To exit deepsleep mode on E290, press and hold boot button, press rst button, release rst and then the boot button */ 

#include <Arduino.h>
#include <esp_wifi.h>
#include <Wire.h>  
#include <driver/gpio.h>

// Include a font (for later)
#include "Fonts/FreeSerif9pt7b.h"

// This sketch uses: 
// The "Preference / 2.1.0 / Volodymyr Shymanskyy/ https://github.com/vshymanskyy/Preferences" library 
// to store values in eeprom of ESP32 between reboots
#include <Preferences.h>

Preferences prefs;

#define PREFMYNAMESPACE "MyDisplay"

void put_preferences(char *var_name, unsigned int x) {

  prefs.begin(PREFMYNAMESPACE); //namespace
  prefs.putUInt(var_name, x);
  prefs.end();
}

unsigned int get_preferences(char *var_name){
  unsigned int x = 0;
 
  prefs.begin(PREFMYNAMESPACE); //namespace
  x = prefs.getUInt(var_name,0);
  prefs.end();

  return x;
}

unsigned int update_counter_clean(const int d)
{
 char cname[] = "recounter";
 unsigned int recounter = get_preferences(&cname[0]);
 
  // on the very first time the counter can be undefined
  if(recounter <0 || recounter> (int)(3600/d) ) recounter = 0;

  if(recounter == (int)(3600/d) ){
      recounter = 1;
      display_clear();
  } else {
      recounter++;
      display_fastclear();
  }

  put_preferences(&cname[0], recounter);

  return  recounter;
}

/* --------------------------------------------------------- */

void gotoDeepSleep(const uint32_t seconds) {

  // power down other chips
  esp_wifi_stop();
  SPI.end();
  Wire1.end();
  
  // Configure wake sources
  esp_sleep_enable_timer_wakeup((seconds * 1000000ULL)  - (millis() * 1000ULL) ); // Timer wake

  // Enter deep sleep
  esp_deep_sleep_start();
}

/* ----------------------------------------------- */

void display_start(void){
    if (esp_sleep_get_wakeup_cause() != ESP_SLEEP_WAKEUP_TIMER)  display_clear();

    display.fastmodeOn(false);  // false to disable memory-init
    display.landscape();
    display.setTextColor(BLACK);
    display.setBackgroundColor(WHITE);
    display.setWindow( 0,0, display.width(), display.height() ); 
    display.fastmodeOn();
  }

void display_text(int x, int y, char* string) {
  display.setFont( &FreeSerif9pt7b );
  display.setTextSize(2);
  display.setCursor(x, y);
  display.println(string);
}

void display_fastclear(void){
  display.clearMemory();
  display.setBackgroundColor(WHITE);
  display.fillRect(0, 0, display.width(), display.height(), WHITE);
  display.update();

  display.clearMemory();
  display.setBackgroundColor(WHITE);
  display.fillRect(0, 0, display.width(), display.height(), WHITE);
  display.update();

  display.clearMemory();
  display.setBackgroundColor(WHITE);
  display.fillRect(0, 0, display.width(), display.height(), WHITE);
  display.update();

  display.clearMemory();
  display.setBackgroundColor(WHITE);
  display.setTextColor(BLACK);
}

void display_update(void){ display.update(); }

void display_stop(void){ display.fastmodeOff(); }

void display_clear(void){ display.clear(); }

/* ------------------------------------------------*/ 

void setup() {
  unsigned int  counter = 0;
  char          scounter[16];
  char          str[] = "Hello: ";
  const int     xwait = 10;

  display_start();  

  counter = update_counter_clean(xwait);

  display_text(10,70,&str[0]);
  sprintf(&scounter[0],"%d",counter);
  display_text(150,70,&scounter[0]);
  display_update();

  display_stop();  

  gotoDeepSleep(xwait);  

}

void loop() { }
```
 

Greetings Andreas

todd-herbert, 2025-05-05:
@bohnelang
> May I ask for a code review of the void display_fastclear(void) part?

It's important to understand that "fast clear" doesn't heal the display. It causes the same wear as any other fastmode update. I think probably you can remove it entirely. You should be able to change directly from one image to the next (if I remember correctly)

Also, it looks like this code runs every 10 seconds, right? And only one normal update per hour? That's 360 fast updates per normal update, which is *a lot*. My experience is that VM-E290 can only do < 10 fast updates in a row. More than this seems to be bad for this display.

I notice you that you update the display every time the device wakes. If possible, I suggest checking first if your data has changed, and only updating if it is different. There may be other ways to design your program to reduce the number of updates you need. If you are creative, 10 updates can last a long time.

If you already did a lot of fast updates, and your display seems very unhealthy, you can probably help it by using a script to do a lot of updates with fastmode off. Maybe one every 30 seconds, for many hours.

bohnelang, 2025-05-05:
Thank you for your long and comprehensive answer. The 10 seconds are only for the this example to have a counter effect. While monitoring my photovoltaic inverter I am using a refreshing time of 30 seconds. My VM-E290 is running for days and it does not look unhealthy. The white filled rect seems to have an effect because the ghost artefact pattern are strongly reduced (in contrast to a work without). And during the three times of my "fast_clear" you can see that black pattern are vanishing from filled rect to next filled rect.  

[image]

todd-herbert, 2025-05-05:
I don't think I'd be comfortable using that technique myself, but I couldn't say for sure whether you'll have any issues with it or not.

One interesting thing I notice: the screen protector has a green tab. I haven't seen that on an a VM-E290, so it sounds like this might be a new batch of displays.