todd-herbert/heltec-eink-modules #12: Compiling

vinouz, 2024-04-06:
Hello, 

When I try to compile the e-ink module (Wireless Paper with LCMEN2R13EFC1 display), I always get the compile error message : 

/Users/B/Documents/Arduino/libraries/heltec-eink-modules/src/Platforms/WirelessPaper/power_controls.cpp: In function 'void Platform::prepareToSleep()':
/Users/B/Documents/Arduino/libraries/heltec-eink-modules/src/Platforms/WirelessPaper/power_controls.cpp:81:9: error: 'gpio_hold_en' was not declared in this scope; did you mean 'gpio_mode_t'?
   81 |         gpio_hold_en((gpio_num_t) PIN_LORA_NSS);    // "stay where you're told"
      |         ^~~~~~~~~~~~
      |         gpio_mode_t

Is there a way to solve this ?

todd-herbert, 2024-04-07:
I just checked now, and I can't reproduce the error here.

My first thought is that you may not be compiling for the right board. 
Can you give me a bit more information about your environment? 
* Are you using the Arduino IDE, or PlatformIO? (both acceptable)
* If using the Arduino IDE, have you selected Wireless Paper from the drop-down menu?
* If using PlatformIO, could you post your `platformio.ini` file?

If you're not yet familiar with the tools, it could be helpful to take a look at the [getting started](https://github.com/todd-herbert/heltec-eink-modules/blob/main/docs/WirelessPaper/wireless_paper.md#getting-started) page. It has step by step instructions for setting up your environment.

todd-herbert, 2024-04-07:
Thank you for bringing this to my attention. It seems that the latest firmware released by Heltec (v3.0.0-alpha) has removed the `gpio_hold_en()` method. I will look into an alternative. 

In the meantime, you should be able to downgrade to the previous release (v1.0.1) by change the url in the Arduino IDE board manager to `https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/releases/download/1.0.1/package_heltec_esp32_index.json`

todd-herbert, 2024-04-07:
Thank you for bringing this to my attention. It seems that the latest firmware released by Heltec (v3.0.0-alpha) has made some changes. I will look into this now.

In the meantime, you should be able to downgrade to the previous release (v1.0.1) by change the url in the Arduino IDE board manager to https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/releases/download/1.0.1/package_heltec_esp32_index.json

todd-herbert, 2024-04-07:
It seems that there may be significant changes and/or bugs in Heltec's version 3.0.0-alpha. I will continue to investigate this in the future. 

Arduino IDE users should continue to use v1.0.1 of `Heltec ESP32 Series Dev Boards` from the board manager.
The [Getting started with Wireless Paper](https://github.com/todd-herbert/heltec-eink-modules/blob/main/docs/WirelessPaper/wireless_paper.md#arduino-ide) guide has been updated to mention this.

PlatformIO users can continue as usual.

todd-herbert, 2024-04-12:
Hopefully all fixed with [v4.3.0 release](https://github.com/todd-herbert/heltec-eink-modules/releases/tag/v4.3.0), available shortly in Arduino IDE. Any issues, please let me know!

Glenno-H, 2024-04-16:
Hi Todd, 

I think I may still be getting the error with Heltec 3.0.0 Firmware, Heltec ESP32 Dev-Boards Library and eink 4.3.0 when trying to use LORA and the eInk Library at the same time.

Please let me know if there is any information I can provide to asisst.  Thanks!

o:\OneDrive\Projects\Microcontroller\C\libraries\heltec-eink-modules\src\Platforms\WirelessPaper\power_controls.cpp: In function 'void Platform::prepareToSleep()':
o:\OneDrive\Projects\Microcontroller\C\libraries\heltec-eink-modules\src\Platforms\WirelessPaper\power_controls.cpp:80:9: error: 'gpio_hold_en' was not declared in this scope; did you mean 'gpio_mode_t'?
   80 |         gpio_hold_en((gpio_num_t) PIN_LORA_NSS);    // "stay where you're told"
      |         ^~~~~~~~~~~~
      |         gpio_mode_t
Multiple libraries were found for "SD.h"
  Used: C:\Users\sirgo\AppData\Local\Arduino15\packages\Heltec-esp32\hardware\esp32\3.0.0\libraries\SD
  Not used: C:\Users\sirgo\AppData\Local\Arduino15\libraries\SD

Using library heltec-eink-modules at version 4.3.0 in folder: O:\OneDrive\Projects\Microcontroller\C\libraries\heltec-eink-modules 
Using library SPI at version 2.0.0 in folder: C:\Users\sirgo\AppData\Local\Arduino15\packages\Heltec-esp32\hardware\esp32\3.0.0\libraries\SPI 
Using library SD at version 2.0.0 in folder: C:\Users\sirgo\AppData\Local\Arduino15\packages\Heltec-esp32\hardware\esp32\3.0.0\libraries\SD 
Using library FS at version 2.0.0 in folder: C:\Users\sirgo\AppData\Local\Arduino15\packages\Heltec-esp32\hardware\esp32\3.0.0\libraries\FS 
Using library Heltec ESP32 Dev-Boards at version 2.0.1 in folder: O:\OneDrive\Projects\Microcontroller\C\libraries\Heltec_ESP32_Dev-Boards 
Using library Wire at version 2.0.0 in folder: C:\Users\sirgo\AppData\Local\Arduino15\packages\Heltec-esp32\hardware\esp32\3.0.0\libraries\Wire

todd-herbert, 2024-04-16:
@Glenno-H That's definitely got me a bit confused! Hopefully can figure out why the `gpio_hold_en()` method isn't found in this case.

Just gathering some more information:
* What version of the Arduino IDE are you using?
* Which board type is selected in the Arduino IDE?

It might help to "clean" the project, to force everything to rebuild. In Arduino IDE, one way to do this is to change to a different board type (such as Arduino Uno), attempt to build (will likely fail), then swap back to the correct board and try again.

The `gpio_hold_en()` method which is missing *should* automatically be found. If the issue remains after "cleaning the project", you could try manually adding the line `#include <driver/gpio.h>` at the very start of your sketch. Please do let me know the result of adding this line.

todd-herbert, 2024-04-16:
Oh, I *can* see an error actually when using Heltec's LoRa library. I think I missed this one because I personally prefer the RadioLib library for LoRa.

I'll see if there's an easy fix for the conflict.

Glenno-H, 2024-04-16:
Thanks heaps for your quick reply Todd!  Really Appreciate it!

Arduino IDE Version : 2.3.2
The board is the "Wireless Paper" from the Heltec ESP32 Dev-Boards.

I tried the cleaning trick you suggested plus the include option, and sadly they didnt work.  Im very new to Lora, and didnt know about "RadioLib"....Ill check that out.  Thanks!

[image]
[image]

todd-herbert, 2024-04-16:
Thanks for following up with that info!

I'm pretty sure I've spotted the issue now: a file in the "Heltec ESP32 Dev-Boards Library" has the same name as a file in the "Heltec 3.0.0 Firmware". When asking for this file, we are getting the file from the library, rather than the file we want from the firmware.

I've added a quick work-around which seems to distinguish between the two. It doesn't *seem* to have break anything..

Let me know if v4.3.1 sorts the issue for you. It'll probably take a few hours to show up in the Arduino IDE library manager, or you can [grab it right now as a zip](https://github.com/todd-herbert/heltec-eink-modules/releases/tag/v4.3.1)

Glenno-H, 2024-04-16:
Works perfect!  Thanks again!

eiiot, 2024-08-29:
I'm still running into this issue on v4.4.0, sadly:

```
/Users/eliot/Library/Arduino15/packages/Heltec-esp32/tools/esp-xs3/2302/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/dk/8rqvrn_96f77jx0cnj8ndvgr0000gn/T/arduino/cores/8b1972ccdc0c591f4eba9b4bb6647baa/core.a(main.cpp.o):(.literal._Z8loopTaskPv+0xc): undefined reference to `_Z4loopv'
/Users/eliot/Library/Arduino15/packages/Heltec-esp32/tools/esp-xs3/2302/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/dk/8rqvrn_96f77jx0cnj8ndvgr0000gn/T/arduino/cores/8b1972ccdc0c591f4eba9b4bb6647baa/core.a(main.cpp.o): in function `_Z8loopTaskPv':
/Users/eliot/Library/Arduino15/packages/Heltec-esp32/hardware/esp32/3.0.2/cores/esp32/main.cpp:64: undefined reference to `_Z4loopv'
collect2: error: ld returned 1 exit status
Multiple libraries were found for "SD.h"
  Used: /Users/eliot/Library/Arduino15/packages/Heltec-esp32/hardware/esp32/3.0.2/libraries/SD
  Not used: /Users/eliot/Library/Arduino15/libraries/SD
exit status 1

Compilation error: exit status 1
```

closed 2024-08-29 by todd-herbert