todd-herbert/heltec-eink-modules #16: Issues compiling 4.4.0

todd-herbert, 2024-08-29:
I'm still running into this issue on v4.4.0, sadly:

```
/Users/eliot/Library/Arduino15/packages/Heltec-esp32/tools/esp-xs3/2302/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/dk/8rqvrn_96f77jx0cnj8ndvgr0000gn/T/arduino/cores/8b1972ccdc0c591f4eba9b4bb6647baa/core.a(main.cpp.o):(.literal._Z8loopTaskPv+0xc): undefined reference to `_Z4loopv'
/Users/eliot/Library/Arduino15/packages/Heltec-esp32/tools/esp-xs3/2302/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/dk/8rqvrn_96f77jx0cnj8ndvgr0000gn/T/arduino/cores/8b1972ccdc0c591f4eba9b4bb6647baa/core.a(main.cpp.o): in function `_Z8loopTaskPv':
/Users/eliot/Library/Arduino15/packages/Heltec-esp32/hardware/esp32/3.0.2/cores/esp32/main.cpp:64: undefined reference to `_Z4loopv'
collect2: error: ld returned 1 exit status
Multiple libraries were found for "SD.h"
  Used: /Users/eliot/Library/Arduino15/packages/Heltec-esp32/hardware/esp32/3.0.2/libraries/SD
  Not used: /Users/eliot/Library/Arduino15/libraries/SD
exit status 1

Compilation error: exit status 1
```

_Originally posted by @eiiot in https://github.com/todd-herbert/heltec-eink-modules/issues/12#issuecomment-2316538861_

todd-herbert, 2024-08-29:
@eiiot I think this might be something different than issue #12; very keen to catch it though!
I'm not immediately able to reproduce this one here. Can I ask for a bit more info?
* Which device are you using?
* Are you able to share the sketch which is producing the error?

eiiot, 2024-08-29:
I'm using v1.1. My code was just the tutorial code, I think?

```cpp
#include "heltec-eink-modules.h"

EInkDisplay_WirelessPaperV1_1 display;

void setup() {
    display.print("Hello, World!");
    display.update();
}
```

todd-herbert, 2024-08-29:
Ah, Arduino also insist on having a `loop()` method, even if it's empty.
```cpp
#include "heltec-eink-modules.h"

EInkDisplay_WirelessPaperV1_1 display;

void setup() {
    display.print("Hello, World!");
    display.update();
}

void loop() {
}
```

I'll update the readme to make that clearer!

eiiot, 2024-08-29:
That fixed it, thanks!

closed 2024-08-30 by todd-herbert