todd-herbert/heltec-eink-modules #19: `GFX::invertDisplay` disabled / not working

StevenCellist, 2025-04-20:
Hi Todd,

Really happy with this library/interface - it works like a charm. The only thing I now ran into is that `invertDisplay` is disabled as private. I tried setting it as public, but that didn't help. Is there any option for e.g. the VME213 to use inverted display colours? That way I wouldn't need to 'rewrite' the display memory, but could simply invert and display. 

Happy to hear from you, and best regards from @bns on the Heltec forum.

StevenCellist, 2025-04-21:
Hey Todd,

The tangled mess (which I uncovered while tracking down the source of `invertDisplay`) is not at all visible to the end-user, so kudos for that 😉 it works pretty flawless! It's the only solution that I trust for the Heltec e-ink modules...

Judging by your remarks, I'd definitely look at software-invert only. Software-invert is completely fine by me, it just felt like a waste of time and power to redraw everything from sketch... so if there's a simple RAM-invert option, yes!
Bonus points if it allows inverting only part of the RAM.. I'm suffering from feature-creep and would like to 'divide' the screen into two parts, that can be inverted individually. But that's really a bonus item, I'd be happy with a full-screen invert too!

The application is quite a fun one: I'm a teacher (which you may remember) and at school we'd like to use smart 'stickers' during exam week for seating maps for all students - the fixed seating is for making sure everyone is where they're supposed to be and keep track of absence. Which is sooo much work to prepare, distribute, collect, rinse & repeat for all exams for ~900 students over and over again. The inverting is used when a student presses a button on the 'sticker' to check themselves in for presence registration. So it's easy for teachers to verify that students have registered themselves by checking whether the display is white or black - it starts white (with name and ID etc in black text), and when they press the button, invert.

todd-herbert, 2025-04-22:
I've kludged together an invert method which hopefully will do the trick.
https://github.com/todd-herbert/heltec-eink-modules/blob/main/docs/API.md#invert

Calling `invert()` will invert the colors of the full display area. 
Optionally, you can invert a specific region with `invert(left, top, width, height)`.

Inverting a region is not particularly efficient. It's a bit of a hack, making use of the existing graphics code to calculate a mask. It may not be elegant, but it was quick to write and didn't force me to actually tidy anything up! Bench-marked on the VM-E213 @ 240MHz, `invert()` takes 0.25ms, with the equivalent `invert(0, 0, display.width(), display.height())` taking 15ms. This isn't great, but I don't imagine a few extra milliseconds are particularly critical here.

___ 

I think I do understand the general idea of your project now, thank you!

> it just felt like a waste of time and power to redraw everything from sketch

I understand where you're coming from, but I also think you probably don't need to worry too much about the time / power used to regenerate the image from sketch. I would think it would be fairly insignificant compared to the time / power used in the process of physically refreshing the display.

closed 2025-04-23 by StevenCellist

todd-herbert, 2025-04-23:
Do keep in mind that it's maybe 50ms to render, but ~3500ms for full refresh, so feel free to go crazy with the rendering!

StevenCellist, 2025-04-24:
Most of the updates happen with a fast refresh, so in that case it does make more of a difference. And it adds up over hundreds of refreshes after all ;) but you are very right, it is definitely not a crucial factor in power consumption.

StevenCellist, 2025-04-24:
@todd-herbert any chance you can release 4.5.0 on PlatformIO as well?

todd-herbert, 2025-04-24:
Oh yep I'll do that too!