todd-herbert/heltec-eink-modules #20: Gxepd - Support - Vision Master

parahooners, 2025-05-17:
Has anyone use the GxEpd lib with the E213 Vision Master ? apricate its an older lib but I have a few project that already run on this that I'm porting over.

Thanks 

M

todd-herbert, 2025-05-18:
Not the original GxEPD as far as I'm aware. As for GxEPD2, the Meshtastic project used to use a fork which had basic support for the E213's display  although it wasn't even intended for wider use, so proceed with caution.
https://github.com/meshtastic/GxEPD2/blob/master/src/epd/GxEPD2_213_FC1.h

todd-herbert, 2025-05-18:
If you're up for the challenge, it's probably possible to backport the display to the original GxEPD library.

I wrote up a bit of a general overview of the steps here: https://forum.arduino.cc/t/lcmen2r13efc1-discussion-heltec-wireless-paper/1211430

I think some of those configuration values have been refined or corrected since that initial post, so it'd pay to cross-reference it with the driver here:
https://github.com/meshtastic/firmware/blob/master/src/graphics/niche/Drivers/EInk/LCMEN2R13EFC1.cpp