todd-herbert/heltec-eink-modules #21: SX1262 can't be initialized on Wireless Paper after deep sleep

mschorer, 2025-06-08:
On the wakeup after a deep sleep, my sketch fails to find the SX1262
```
11:22:25.453 -> init ... error [fffffffe] - postpone.
11:22:35.549 -> ERR_CHIP_NOT_FOUND
```
if im using:
```
  Platform::prepareToSleep();
  if (seconds) {
      esp_sleep_enable_timer_wakeup((uint64_t) 1000 * 1000 * seconds);  // Microseconds
  }
  // Sleep now
  esp_deep_sleep_start();
```
If i skip the "// Set SX1262 to SLEEP mode..." part in prepareToSleep, the epaper finds the chip.

But - i still have to do a hw reset after an upload.

Any ideas?

todd-herbert, 2025-06-11:
Hmm, if I was guessing, it might be that the LoRa radio's CS pin is still held.

Maybe try adding this line early in setup:
```cpp
gpio_hold_en((gpio_num_t) PIN_LORA_NSS);
```

Let me know if it helps, and I can modify the library to handle it automatically.

mschorer, 2025-06-11:
> Maybe try adding this line early in setup (before LoRa init):

Hmmh, this is (identical to) the last command in prepareToSleep ... where
it pins the gpio high during deepsleep afaics.

Wouldn't i need to "undo" this before radio.begin, so the CS can go
(active) low?

I will read up on and play wit the idea .... maybe not today ...

Thanks for the direction ...

mschorer, 2025-06-11:
Hah! That seems to do it! 

Will report on how it did overnight tomorrow ...

mschorer, 2025-06-12:
Yep, working through the night.
It looks like, but not 100% sure for the further reduced power consumption yet.

closed 2025-06-12 by mschorer