todd-herbert/heltec-eink-modules #22: Heltec Wireless Paper new variant?

alexsavuto, 2025-06-14:
"Hello, I’m having an issue with what I believe is a new variant of the device. The PCB is the same as version 1.1, but there’s a sticker labeled 'V1.1.1' inside the flex ['HINK-E0213A162-A1' 'DEN2512']. It arrived with the screen working, but I haven’t been able to get it to function properly—the only thing I’ve managed is running the 'snowflakes' demo by uploading the example .ino file for V1. Has anyone seen anything about this? I also tried the Meshtastic firmware (both V1 and V1.1), but I still can’t get the screen to work."

[image]
[image]
[image]

todd-herbert, 2025-06-15:
Yes, this is a new hardware revision that Heltec have begun shipping. I'm waiting on delivery here, but once it arrives I'll get it set up with Meshtastic and add the driver here too.

alexsavuto, 2025-06-15:
I'm glad to hear that. Likewise, I've emailed Heltec to request documentation about this variant. Yesterday, I spent several hours testing different screen drivers to see if any matched based on the V1.1 pinout, but no luck. So, I'll just have to wait for new updates, as my knowledge has reached its limit. Thank you very much, and if you need me to test anything, let me know.

alexsavuto, 2025-06-29:
nop, only work the new oficial example a im waiting heltec-eink-modules.h update, my knowledge is too limited to do the integration.

todd-herbert, 2025-06-29:
It looks like my v1.1.1 board got lost in the mail.. slows things down a bit

closed 2025-07-13 by todd-herbert